arduino-libraries/ArduinoMotorCarrier #2: Please provide better instructions on how to flash the firmware

phretor, 2018-07-04:
I'm testing the PID control and the motors act weird. I'd like to make sure it's not the (old) firmware's fault (I'm running 0.0.6), but it's not very clear how to flash the carrier board.

facchinm, 2018-07-04:
Hi @phretor,
firmware 0.0.6 has some bugs that has been solved in the meantime; to flash the a new firmware you simply need to:
- make sure that library is updated (1.0.3 right now)
- open the `Flasher` example 
- upload it on the user-programmable MKR board (MKR1000 for example)
- briefly remove the power from both the Motor carrier and the user board
- open the MKR1000 serial port with any program (like Arduino IDE serial monitor) 
- the new firmware will be automatically flashed and the new fw version will be reported if the flash completed successfully.

Let me know if the procedure works fine!

phretor, 2018-07-05:
Hi @facchinm,

thanks for the quick answer. The flashing procedure did work: I'm now happily running 0.0.8. At some point I'll fork the repo and convert this issue into a wiki page as a courtesy to the next passenger.

I'm still experiencing a weird PID control action on the motors: not sure if it's the reading from the encoders that are not accurate, or the actuation. I've tried various Kp, Ki, Kd combos, with no significant improvements.

I'm using the DC motors with encoder that ship with the Arduino Engineering Kit, hooked up as `M1` and `M2`, and encoders to Hall sensors HA1-HB1 and HA2-HB2, respectively. I've inverted the +/- so that a set point to, say, position X would mean "go forward" for both motors, and inverted the polarity of the encoder wirings accordingly.

Here's the code I'm playing with. I think you want to look at the `moveBoth` function, as the rest is mostly MQTT command parsing and connectivity:

```Arduino
#include <WiFi101.h>
#include <SPI.h>
#include <PubSubClient.h>

#include <MKRMotorCarrier.h>

#define PID_P 10.0
#define PID_I 0.0
#define PID_D 3.0
#define MIN_FORK 9
#define MAX_FORK 100
#define MAX_ACCEL 200
#define SPEED 25
#define MIN_SPEED 25
#define MAX_SPEED 70
#define PREFIX "/rover/command"

char ssid[] = "************";
char pass[] = "***********";

int status = WL_IDLE_STATUS;
float kp = PID_P;
float ki = PID_I;
float kd = PID_D;

WiFiClient wifiClient;
PubSubClient client(wifiClient);
IPAddress server(192, 168, 42, 225);

void commandSetGainP(float g) {
  kp = g;
}

void commandSetGainI(float g) {
  ki = g;
}

void commandSetGainD(float g) {
  kd = g;
}

void moveBoth(int target1, int target2) {
  Serial.print("Move to target: ");
  Serial.print(target1);
  Serial.print(", ");
  Serial.println(target2);

  // motor 1 = right
  // motor 2 = left
  
  encoder1.resetCounter(0);
  encoder2.resetCounter(0);
  
  pid1.setControlMode(CL_POSITION);
  pid2.setControlMode(CL_POSITION);

  pid1.setMaxAcceleration(MAX_ACCEL);
  pid2.setMaxAcceleration(MAX_ACCEL);

[... 255 lines cut ...]

void loop() {
  if (!client.connected()) {
    reconnect();
  }

  Serial.print(encoder1.getRawCount());
  Serial.print(" ");
  Serial.print(encoder2.getRawCount());
  Serial.print("\n");

  client.loop();
}
```

Here's a video that gives an idea of the issue I'm talking about, when issuing two commands:

1. go to 1000 position
2. go to -1000 position

Video is hosted here: https://www.dropbox.com/s/1sfr8br4hew7sz7/IMG_0307.MOV?dl=0

As you can see, the motors move a bit erratically, as if under the effect of bad control. If I simply set the duty cycle of the motors to, say, 50%, they will run smoothly. So, the erratic movement must be part of the control.

I can try with different PID parameters or do other tests, and I'm happy to contribute with the full working example once completed.

facchinm, 2018-07-05:
Glad the firmware update worked smoothly!
I think the PID problem might be related with the missing call to `controller.ping()` .
Regularly calling that function is compulsory (at least once every three seconds), otherwise the motors will stop (this was added as a safety measure in case the controlling board stops responding). Simply add it before `client.loop();` and it should be enough

phretor, 2018-07-05:
Actually, nothing really changed. I had it there yesterday (not sure how it got removed), each 100ms.

I'm now calling it each 1000ms, but the twitching movement is still there.

facchinm, 2018-07-05:
@ErnestoELC does it look familiar to you? Could you try to reproduce the issue?

ErnestoELC, 2018-07-06:
Hi @phretor 
I was able to reproduce the shaky behavior of the motors when using the PID. I am not sure what is the reason for it yet. It might be the way the PID is implemented or that value of the gains are not properly tuned.
We'll dig a bit more and come back to you. 
Thanks!

phretor, 2018-07-06:
Thanks for testing this out @ErnestoELC - makes me feel less lonely :-)

Should you need any tests on my side, just shoot a message here or in private.

facchinm, 2019-01-16:
This should be fixed with latest 0.10 firmware. Reopen if needed

closed 2019-01-16 by facchinm