arduino-libraries/ArduinoMotorCarrier #10: M1.setDuty not sending voltage

malk0lm, 2019-01-15:
I have a DC motor hooked up to M1 but when I try to setDuty(100) nothing happens. I'm using the example sketch Motor_test however I removed the battery code, since I'm powered by the USB.

I see the Serial output, but I don't see any movement of the motor. When I jump the motor to the 3.3v pin it works fine. I've previously tested it with a 5v pin and that works fine too.

Any ideas? Thanks!

facchinm, 2019-01-16:
Hi @malk0lm ,
M1.setDuty simply starts a PWM on the pin (see https://github.com/arduino-libraries/MKRMotorCarrier/blob/master/src/DCMotor.cpp#L56). 
The entire board must be powered with more than 9 volts otherwise it's not going to work correctly (thus the battery code). @ErnestoELC is it correct?

malk0lm, 2019-01-16:
Hi @ErnestoELC would a 5VDC wall powered converter work? It sounds like I could use some kind of barrel jack connected to the 5V PWRIN?

Thanks again!

ErnestoELC, 2019-01-17:
Hi @malk0lm ,
The recommended voltage input for the MKRMotorCarrier is from 6.5V to 15V. 
Hope this helps!

malk0lm, 2019-01-17:
Hi @ErnestoELC,

Thanks so much for the help! Just one more thing... I'm not so good with the hardware, but my understanding is that the input is 5V, wouldn't putting more than 5V into that power circuit run the risk of shorting out some of the components that are connected to that?

Sorry for the dumb question.

Thanks again!

ErnestoELC, 2019-01-18:
Hi @malk0lm ,
You can connect a 3 cell (11.1V) LiPo battery to the power input of the motor carrier. The motor carrier has an internal switch voltage regulator that will convert that voltage to 5V. Those 5V are used to power the MKR1000 and the servo motors. The DC motors are powered by the Lipo directly (11.1 V) via the motor drivers in the board. The carrier also has a 3V output. This 3V are coming from the regulator inside the MKR1000.
Hope this helps!

malk0lm, 2019-01-26:
Thanks! With your advice I got it to work!

closed 2019-01-26 by malk0lm