arduino-libraries/ArduinoMotorCarrier #11: What is the velocity unit ?

Bug38, 2019-01-29:
Hi,

I'm trying to use the PID velocity mode but I can't figure which unit it uses...
I'm working with a motor which has a 12 poles encoder, 88 ration gearbox reducer and I try to set it turn at 0.63 turn per second.
I saw that encoders count each change of the two sensors, so I figured a 3696 ticks/seconds velocity.

BUT, when I set-up the pid, I see that it turns WAY too fast and by printing `encoder1.getCountPerSecond()` I can read `63`.

Have you an answer or some tips for me ?

Thanks

Bug38

facchinm, 2019-01-29:
@ErnestoELC can you take a look?

ErnestoELC, 2019-02-13:
Hello Bug38,
The reason for the last timestamp is because the call to populateRegisters_wrapper() is rescheduled every X ms defined when you instantiate the PID:
`PIDWrapper(encoders[0].position, encoders[0].velocity, &dcmotors[0], 0, 10, 100), //10ms period velo, 100ms period pos .`

So basically as it is right now, depending on how frequent you set this eventCall you will get a different unit for countsPerSecond. The default unit is counts/10ms.

Hope this helps!

closed 2020-06-22 by facchinm