arduino-libraries/ArduinoMotorCarrier #13: J1-J4 connector types?

malk0lm, 2019-02-18:
Hi again, 

Sorry I'm not very good when it comes to hardware. I'm a software person. Can you please tell me what the connector types are for IN1-IN4 I think they are labelled J1-J4.

I thought it was a JST type but I don't know what kind.

Thanks for the help!

malk0lm, 2019-02-21:
Maybe its a 3-pin fan connector? I'm really struggling trying to find something that will work.

ErnestoELC, 2019-02-22:
Hi @malk0lm,

The component is the LHA-03-TS from Adam Tech. There are other connectors that are similar and are probably compatible with this one like the 640456-3 from TE connectivity or some of the JSTs

Hope this helps. 
E.

malk0lm, 2019-02-22:
Thank you Ernesto! You have been extremely helpful while I try to figure out how to use this MKR board!

closed 2019-02-22 by malk0lm