arduino-libraries/ArduinoMotorCarrier #16: change the PWM frequency on a MKRMotorCarrier

norberto-pires, 2020-02-02:
Hello, Is there an implementation on setting the PWM frequency for the servos in a MKRMotorCarrier? The library has the function but not implemented.
Thanks.
norberto

facchinm, 2020-02-17:
Hi @norberto-pires ,
there is not so much space left on the D11 microcontroller to add this feature, but for M3 and M4 (controlled by the D21) there is a chance (https://github.com/arduino-libraries/MKRMotorCarrier/blob/master/src/DCMotor.cpp#L39). If anyone's willing to implement it I'll be happy to merge :slightly_smiling_face:

pr3dat0r2201, 2020-03-13:
Hello,
Can I use the PWM frequency library for changing the frequency?
Thanks
Arunaabh Singh

pr3dat0r2201, 2020-03-13:
Hello,
Can I use the PWM frequency library?
Thanks.
Arunaabh Singh

AsherThomasBabu, 2020-03-15:
@facchinm if you point me in the direction, I am willing to give it a try.
Thank you
Asher