arduino-libraries/ArduinoMotorCarrier #24: PID Position Controller does not work

Ravenspark, 2020-05-28:
Hello everyone, hello @facchinm

We have started with example [MKRMotorCarrier Test](https://github.com/arduino-libraries/MKRMotorCarrier/blob/master/examples/Test/Test.ino), however the motor has a really shaky behaviour as described in #2 .

We updated the firmware on the Motor Carrier to the newest (0.11) but the behaviour is still the same.

Additionally we tried to compile the firmware of the Motor Carrier, but the instructions in #8 are not very clear to me. I'm not too familiar with the arduino IDE and how to manually setup new boards...

Any help is greatly appreciated :)
Best Regards
Simon

facchinm, 2020-06-22:
Hi @Ravenspark ,
we released version 2.0 of the library with fw 0.20 that should fix the shaky behaviour.
Can you try updating both the library and the carrier (using Flasher sketch) and report back if it's fixed on your setup? Thanks

FreeSprung, 2022-09-11:
Hello everyone,

I am new to the forum and would like to revisit the topic.

I would like to use the PID function.

I am using a MKR Motor Carrier with a MKR WIFI 1010 and a Nano Motor Carrier with an Arduino Nano 33. I also started with the MKRMotorCarrier Test example and am using a DC servo motor with a 200 lines integrated encoder. Both motor carriers are updated to firmeware 0.20 (according to Serial Monitor).

The motor reacts the same with both boards. It turns in both directions alternately. No control behavior is visible, everything seems chaotic. The encoder signal was checked with an oscilloscope and is very good. The board also detects the motor position correctly.

If I remove the encoder (A and B) or set the encoder position as a constant in the sketch, the motor behavior does not change. The direction of rotation still changes alternately, which makes no sense.

If I set the PID values all to 0 or perform a reset (controller.reboot();), this also has no effect.

In short: The PID1/PID2 commands do not work, the M1/M2/M3/M4 commands do. I can only imagine that it is due to the motor carrier (hardware or software).

With an Arduino Uno and a suitable Motor Shield Rev3 the motor position can be controlled. However, the sampling frequency for the encoder at full speed is unfortunately not high enough.

I get no progress since two weeks and a MKR WIFI 1010 broke during multiple updates.

Can anyone help?

Best regards
FreeSprung

per1234, 2022-09-11:
Closing due to lack of response to the request for feedback made two years ago:

> Can you try updating both the library and the carrier (using Flasher sketch) and report back if it's fixed on your setup? 

@Ravenspark if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

per1234, 2022-09-11:
>Can anyone help?

Hi @FreeSprung. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

closed 2022-09-11 by per1234

FreeSprung, 2022-09-11:
Thank you for the quick feedback.

I will try the forum.

The flasher sketch was used.

Best regards
FreeSprung