arduino-libraries/ArduinoMotorCarrier #62: Portenta H7 compatibility ?

Wilfried71390, 2024-01-04:
Hi Everyone,
I'm using a MKR motor carrier with a Portenta H7 board. 

Simple question : are we sure that both are compatible?

It's impossible to connect to the carrier, the sketch fails here: "Couldn't connect! Is the red LED blinking? You may need to update the firmware with FWUpdater sketch"
No red LED blinking on carrier (neither on Portenta).

I have no other MKR boards (such as MKR zero or MKR 1010 WIFI) to test the carrier with ...

Any idea ?

Thanks

Wilfried

per1234, 2024-01-05:
A claim of support was added in version 2.0.2 of the library: https://github.com/arduino-libraries/ArduinoMotorCarrier/pull/44. However, I can reproduce the fault reported by @Wilfried71390, using the library's example (e.g., `examples/MKRMotorCarrier/Motor_test`) or this minimal sketch:

```cpp
#include <ArduinoMotorCarrier.h>

void setup() {
  Serial.begin(115200);
  while (!Serial);
  Wire.begin();
  if (controller.begin()) {
    Serial.print("MKR Motor Carrier connected, firmware version ");
    Serial.println(controller.getFWVersion());
  } else {
    Serial.println("Couldn't connect! Is the red LED blinking? You may need to update the firmware with FWUpdater sketch");
    while (1);
  }
}
void loop() {}
```

It works as expected when I use the MKR Zero board instead of the Portenta H7.

Wilfried71390, 2024-01-06:
Hello,
Thanks a lot per1234, i'm feeling less alone!
Why did you change the title of the post in 'controller.getFWVersion'? Does it mean that the issue is link to this request? Does it mean that the 'controller.begin' should work ?
Regards
Wilfried

Wilfried71390, 2024-01-06:
An other question: did you try to update the firmware of the motor carrier (through the MKR zero board) and then make a new try with the portenta?